simonsobs/sodetlib #266: take_bgmap fails with all NaN array encountered

dpdutcher, 2022-07-27:
https://github.com/simonsobs/sodetlib/blob/d62faa64343e4931db5442466ac9aeea23bcecb3/sodetlib/operations/bias_steps.py#L542

Debugging is hard because no object is returned; should catch this and return something more useful.

jlashner, 2022-07-27:
I thought the bias step analysis object is still return and can be further inspected: https://github.com/simonsobs/sodetlib/blob/d62faa64343e4931db5442466ac9aeea23bcecb3/sodetlib/operations/bias_steps.py#L1142-L1152

Is that not the case?

jlashner, 2022-07-27:
Ah maybe its not returning because it's trying to plot the bg assignment: https://github.com/simonsobs/sodetlib/blob/d62faa64343e4931db5442466ac9aeea23bcecb3/sodetlib/operations/bias_steps.py#L1014-L1015

We should check if it actually finished the analysis before plotting

closed 2022-08-10 by dpdutcher