simonsobs/sodetlib #268: NaNs ruin polyfits in analyze_iv

dpdutcher, 2022-08-10:
https://github.com/simonsobs/sodetlib/blob/c1acd700e161d75f9e75980dbbb9a3115e7b373c/sodetlib/operations/iv.py#L364-L368

If iva.resp[i] contains a `nan`, these polyfits will return `[nan. nan]`, and then all of iva.resp[i] becomes `nan`. 
In my testing, I've seen that if one detector has a NaN, then every detector on that bias line has a NaN at that same index. So this issue drops whole bias lines from iv results.

Relatively simple fix, using np.isfinite to index the input arrays to the polyfits. PR incoming.

jlashner, 2022-08-10:
Just approved this PR, but do you have any how NaNs propagate to other channels on the bias line? Is it through the `i_bias` var or `resp`?

dpdutcher, 2022-08-10:
The NaNs enter in when `resp` is being populated from the AxisManager:
https://github.com/simonsobs/sodetlib/blob/c1acd700e161d75f9e75980dbbb9a3115e7b373c/sodetlib/operations/iv.py#L326

I tried changing that to np.nanmean, and that didn't solve the issue, so the time range `m` it's averaging over must be entirely NaNs.

closed 2022-08-10 by dpdutcher

jlashner, 2022-08-10:
ah ok, to be honest I wasn't aware axismanagers even loaded with NaNs

PR #269 (2022-08-10): In iv_analysis, clean input arrays of nans before polyfit
Resolves #268 

I tested this by re-analyzing an old IV, and I recovered a bias line's worth of data that had been dropped.
I believe this is the main culprit of why I was getting different and varying yields from newer sodetlib IVs compared to older versions.