simonsobs/sodetlib #276: All tunefiles report Quality Factor equal to 1 for all resonators.

chw3k5, 2022-08-17:
The every tunefile and every resonator have a Quality Factor equal to 1 (indicating a no resonator and perfect transmission line).

You can see the issue with the follow code snippet.

```
import numpy as np
tune_path = '/mnt/so1/data/ucsd-k2so/smurf/16463/ufm_mv11/1646318795_uxm_relock/outputs/1646320098_tune.npy'
tunefile_data = np.load(tune_path, allow_pickle=True).item()
for smurf_band in sorted(list(tunefile_data.keys())):
    if 'resonances' in data_this_band.keys():
        resonator_data_this_band = data_this_band['resonances']
        available_res_indexes = sorted(resonator_data_this_band.keys())
        for res_index in available_res_indexes:
            if tunefile_data[smurf_band]['resonances'][res_index]['Q'] == 1:
                print(f'Quality Factor is 1 for (SMuRF, res_index): ({smurf_band}, {res_index})')
```

In the tunefile there are arrays of frequency and complex S21 from which the quality factor (and other other parameters) can be determined by model fitting. This in not a desirable situation for two reasons:

1. There is [a standard fitting](https://github.com/simonsobs/sodetlib/blob/0e31ca2bd366423a77e9780d0d11a4decc4dc2d9/sodetlib/resonator_fitting.py#L183) that is provided for, but not applied to, the tunefiles from sodetlib. Any project that wants to use this process than becomes dependant on sodetlib and it's requirements. This complicates installation, execution, deployment, modulability of any project that want to use this existing and maintained tool. This would get my project, DetMap, in trouble as I also resist calls to from sotodlib to integrate into that monolithic repository (put the AxisManager in into own repo and then we can talk).
2. Any modeling (like doing a quality factor fit) is going to be a significant slow down to the processing. For the ~2000 resonators for a single tunefile this is going to demoniate the computational time in DetMap even compared to the 'world's most complicated plots' that represent 95% of the computational time per tunefile currently.

For these reasons, I ask that this Quality factor be calculated and reported correctly with each tunefile. I also request numpy.nan and not 1 to indicate null values. Keeping track of the slang for null, None, nan is across SO data products difficult; Quality factor == 1 look like a model edge case failure and not a did-not-fit situation. I will be amending my other issue reports to reflect this new understanding.

jlashner, 2022-08-18:
Hi Caleb, just a few comments on this.

- Tunefiles are generated by pysmurf, so we don't really have the ability to modify what's in them on creation with just sodetlib. This means that if we do add real resonator parameters to the smurf tunes and save the results, we'll still have two sets of tune files, one with resonator parameters and one without
- Because resonator fitting takes a while and isn't used at all for regular smurf usage, I don't see any strong reason to put the procedure into the standardized "setup" operation, and would rather it just be run in post if/when it's actually needed.

Instead I think the resonator fit function should be run in the analysis pipeline, and the output file should be saved and written to CalDb or some other archive. Then the DetMapping repo can just take resonator-fit files as input and not worry about performing the fit itself. Maybe @kmharrington or @jseibert575  can comment on whether they agree with this.

kmharrington, 2022-08-18:
I agree that we don't want to start editing the tune files. We've had to accept them as pysmurf gives them to us. 

Side note on sotodlib dependence, the sotodlib module never imports DetMap. There is just currently 2 scripts in the site_pipeline folder that imports sotodlib and DetMap. DetMap could easily import tools from the core of sotodlib. 

Does DetMap need resonator fits to make better maps? We already need to figure out how to do more the comprehensive integration for loading bias group maps into the readout/detector mapping when we're running on-the-fly in the field. We can add resonator fits to it while we're at it. The only question I have is are we calling it on the smurf server or part of the first light pipeline scripts.

jlashner, 2022-08-18:
I do think resonator fits will help with mapping, since it will weed out a lot of spurs that aren't real resonators, and will help us group mux-bands based on dip depth. I think it should be run as part of the first-light pipeline, since if it's run on the smurf-servers it will delay data-taking by a non-negligible amount when it doesn't really need to. I recall it taking ~20-30 minutes per tune.

kmharrington, 2022-08-18:
30 minutes?!?! 🤯 

Yeah, we can look at running them on the DAQ nodes / Analysis nodes. Can we also look at optimizing them? That seems way too long for the amount of analysis required for this.

jlashner, 2022-08-18:
We can try to optimize, but it is a 9-parameter fit so I'm not sure there's much we can do on our end to make that faster.

jlashner, 2022-08-18:
Though, one could look at how the fits compare with the initial guesses: https://github.com/simonsobs/sodetlib/blob/0e31ca2bd366423a77e9780d0d11a4decc4dc2d9/sodetlib/resonator_fitting.py#L74-L91

We can probably figure out a way to estimate Q and dip-depth without performing the full fit.

closed 2023-06-29 by dpdutcher