simonsobs/sodetlib #277: Tune files saved to the same directory on the smurf-server

jlashner, 2022-08-26:
Just noticed that all tune-files are being saved to the same directory on the smurf-server. This can be really bad, especially if multiple slots are tuning at the same time, in which case it's possible that one slot overwrites the tune for another.

@dpdutcher do you know if princeton has seen any issues of this with multi-slot streaming?

Easiest way to fix this is probably to change the `tune_path` in the pysmurf-config so it's different for each slot, but that is another thing to keep track of and remember to change. Really these should just go into the pysmurf-output folder, since we're not actively removing those from systems.

dpdutcher, 2022-08-26:
No we haven't seen issues with this, though it has always struck me as a bit strange there isn't at least a stream_id subdivision in that folder like there is for the pysmurf outputs.

kmharrington, 2022-08-26:
I think the big thing is making sure we're still able to find tune files after they've been made. My memory of pysmurf is that things like `S.load_tune()` just looks in the tune folder and loads the most recent one (if the path isn't specified). I might say lets put them in `tune/stream_id/` instead of the dated folders (which is what I think you mean by pysmurf-output folders). Because these are special enough that we want them to be easy to find. 

Especially if we start needing to delete things in the pysmurf-output folders, it will be nice to have a clear distinction of what's easily deleteable and what's not. Tune files and channel assignments are the things that we might want to reference on the smurf server weeks/months after we've made them.

jlashner, 2022-08-26:
This PR should fix the issue when merged: https://github.com/slaclab/pysmurf/pull/721

kmharrington, 2022-10-10:
Looks like the pysmurf issue just got merged, what else needs to be done on our side?

kmharrington, 2022-12-20:
sounds like this is complete now.

closed 2022-12-20 by kmharrington