simonsobs/sodetlib #283: Issue when running S.setup for slot 6

jlashner, 2022-09-06:
Currently getting this failure when running `S.setup()` on slot 6. Has anyone seen this before?
```
|======================================================| 1.3k/1.3k (100.00%)         0s
[ 2022-09-06 15:21:39 ]  Setting up...
[ 2022-09-06 15:21:39 ]  Toggling DACs
[ 2022-09-06 15:21:39 ]  caput smurf_server_s6:AMCc:FpgaTopLevel:AppTop:AppCore:MicrowaveMuxCore[0]:DBG:dacReset[0] 1
[ 2022-09-06 15:21:39 ]  caput smurf_server_s6:AMCc:FpgaTopLevel:AppTop:AppCore:MicrowaveMuxCore[0]:DBG:dacReset[1] 1
[ 2022-09-06 15:21:39 ]  caput smurf_server_s6:AMCc:FpgaTopLevel:AppTop:AppCore:MicrowaveMuxCore[1]:DBG:dacReset[0] 1
[ 2022-09-06 15:21:39 ]  caput smurf_server_s6:AMCc:FpgaTopLevel:AppTop:AppCore:MicrowaveMuxCore[1]:DBG:dacReset[1] 1
[ 2022-09-06 15:21:39 ]  caput smurf_server_s6:AMCc:FpgaTopLevel:AppTop:AppCore:MicrowaveMuxCore[0]:DBG:dacReset[0] 0
[ 2022-09-06 15:21:39 ]  caput smurf_server_s6:AMCc:FpgaTopLevel:AppTop:AppCore:MicrowaveMuxCore[0]:DBG:dacReset[1] 0
[ 2022-09-06 15:21:39 ]  caput smurf_server_s6:AMCc:FpgaTopLevel:AppTop:AppCore:MicrowaveMuxCore[1]:DBG:dacReset[0] 0
[ 2022-09-06 15:21:39 ]  caput smurf_server_s6:AMCc:FpgaTopLevel:AppTop:AppCore:MicrowaveMuxCore[1]:DBG:dacReset[1] 0
[ 2022-09-06 15:21:39 ]  caput smurf_server_s6:AMCc:ReadAll 1
[ 2022-09-06 15:21:45 ]  Waiting 20.00 seconds after...
[ 2022-09-06 15:22:05 ]  Done waiting.
[ 2022-09-06 15:22:05 ]  caget smurf_server_s6:AMCc:SmurfApplication:SmurfVersion
[ 2022-09-06 15:22:05 ]  6.0.0
[ 2022-09-06 15:22:05 ]  caput smurf_server_s6:AMCc:setDefaults 1
[ 2022-09-06 15:22:35 ]  caget smurf_server_s6:AMCc:SmurfApplication:ConfiguringInProgress
[ 2022-09-06 15:22:35 ]  False
[ 2022-09-06 15:22:35 ]  caget smurf_server_s6:AMCc:SmurfApplication:SystemConfigured
[ 2022-09-06 15:22:35 ]  False
[ 2022-09-06 15:22:35 ]  System configuration finished after 30 seconds. The final state was False.
[ 2022-09-06 15:22:35 ]  ERROR : System configuration/setDefaults failed!  Do not proceed!  Reboot or ask someone for help.  You are strongly encouraged to report this as an issue on the pysmurf github repo at https://github.com/slaclab/pysmurf/issues (please provide a state dump using the pysmurf set_read_all/save_state functions).
[ 2022-09-06 15:22:35 ]  Setup failed!
[ 2022-09-06 15:22:35 ]  ERROR : System setup failed!  Proceed at your own risk!
Dumping sodetlib configs to /data/smurf_data/20220906/ufm_uv8/1662477699/config/1662477756
```

dpdutcher, 2022-09-06:
Never seen that one

jlashner, 2022-09-06:
Here are the logs in the smurf-streamer docker:
```
ERROR:pyrogue.Variable.RemoteVariable.AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].12BIT_MODE:'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'
ERROR:pyrogue.Variable.RemoteVariable.AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].12BIT_MODE:Error setting value '0' to variable 'AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].12BIT_MODE' with type UInt2. Exception='Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'
ERROR:pyrogue.Command.BaseCommand.AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].JESD_DDC_config:'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'
ERROR:pyrogue.Command.BaseCommand.AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].Init:'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'
ERROR:pyrogue.Command.BaseCommand.AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].InitAmcCard:'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'
ERROR:pyrogue.Command.BaseCommand.AMCc.LoadConfig:'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'

ERROR: Setting defaults try number 0 failed with: 'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'

Setting defaults from file /tmp/fw/smurf_cfg/defaults/defaults_c03_lb_lb.yml (try number 1)
ERROR:pyrogue.Command.BaseCommand.AMCc.LoadConfig:'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'

ERROR: Setting defaults try number 1 failed with: 'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'

Setting defaults from file /tmp/fw/smurf_cfg/defaults/defaults_c03_lb_lb.yml (try number 2)
ERROR:pyrogue.Command.BaseCommand.AMCc.LoadConfig:'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'

ERROR: Setting defaults try number 2 failed with: 'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'

Setting defaults from file /tmp/fw/smurf_cfg/defaults/defaults_c03_lb_lb.yml (try number 3)
ERROR:pyrogue.Command.BaseCommand.AMCc.LoadConfig:'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'

ERROR: Setting defaults try number 3 failed with: 'Memory Error for AMCc.FpgaTopLevel.AppTop.AppCore.MicrowaveMuxCore[1].ADC[1].CH[0].SYNC_REG at address 0x80590008 Verify Error: Local=0x300x00x00x0. Verify=0xff0x00x00x0. Mask=0xcf0x00x00x0'

ERROR: Failed to set defaults after 4 retries. Aborting!

```

jlashner, 2022-09-06:
Seems like this is the same issue Yuhan saw here: https://simons1.princeton.edu/confluence/display/UD/AMC+C03-A01-14+issue

Yuhan says they sent back to SLAC and found that the issue was that one of the AMC connectors was stressed and kind of broke off the board. We should try to understand how that causes a VerifyError like this and how we can confirm this, but that seems like it's possibly the culprit.

dpdutcher, 2025-10-01:
While reseating a board (the carrier and AMCs separately) may help this issue, it is usually a sign that a system needs to be returned to SLAC.