simonsobs/sodetlib #304: Recognizing fixed tones in archival data

jlashner, 2022-10-13:
We don't have a good way of storing fixed tone channels so we can pick out fixed tones of archival G3 data. I'm not really sure what the best way to do this is.... I kind of think that after tuning we should find fixed tone location and store frequencies and channel info in the tunefile. Or we can have a different fixed-tone info file that we save from sodetlib. I'm open to other suggestions though... I assume we will pretty much always want to run with fixed tones enabled so I think this is important to figure out.

kmharrington, 2022-10-17:
This is a really good question / point that we need a tracking infrastructure here. 

Does SMuRF know which fixed tones it's outputting? Ex: is there a function we can call to ask for which tones have been set? 

We technically base our readout_ids off of the channel assignment files (which are compiled into the Tune files). We could do something similar with fixed tones. This entire topic is probably worth a one-off discussion tbh.

jlashner, 2022-10-17:
After talking with Katie, I think a good way of handling this is to have the `stream_g3` functions write a "fixed_tone_mask" file with the fixed tone channel id's that can always be accessed in that pysmurf action

jlashner, 2022-10-17:
Also add feedback_enable array to monitored vars

kmharrington, 2023-01-10:
Making a note here that the current setup that _doesn't_ track/know about the fixed tones are being built into data packaging checks right now, so when this is completed we'll need to track it through the system.

jlashner, 2023-01-25:
I added the `feedbackEnableArray` to the list of tracked metadata here:
https://github.com/simonsobs/smurf-streamer/commit/75c54b802050cf10208222b45bf03a66187370b5

This _should_ give us all that we need to figure out which tones are fixed vs not in analysis, since the fixed tones will be those where the amplitude scale is not 0, but the feedback_enable is 0. Thinking back on this issue and I think that should be sufficient, and I don't think it makes a ton of sense to add this to another calibration file that needs to be tracked.

I can't quite remember if there were additional issues with this method that we talked about offline which made me think this wouldn't be enough info, but for now I'm pretty happy with that just being the solution.

kmharrington, 2023-01-25:
I think that's probably enough. The feedback enable array makes it into the status dump frame right? Any chance you have some files with these on/off?

jlashner, 2023-01-25:
I believe the feedback_enable_array should be in the p10r2 status dumps, but I don't think we've been running with fixed tones

jlashner, 2023-01-25:
I will try to take some data with fixed tones soon when I have the chance

kmharrington, 2025-11-13:
I'm not sure what the exact implementation is for in sodetlib/pysmurf but I know where binding fixed tones into separate entries in the frames and that sotodlib can load them.