simonsobs/sodetlib #394: overbias_det causing overheating problem on MXC

yaqiongl, 2023-10-31:
Hello,

We recently found that overbias_det in util.py giving us a huge problem that overheats the MXC to a few hundreds of mK or even 1K.
We narrow down the problem to line 857 `set_current_mode(S, bias_groups, 1, const_current=False)` 
I'm not really sure how set_current_mode works since there's something related to relay etc, but once we commented out this line, or change to `set_current_mode(S, bias_groups, 0, const_current=False)` , everything go back to normal.

jlashner, 2023-10-31:
Hi Yaqiong,

Setting to high-curret-mode is pretty necessary for overbiasing dets, since that is what actually pushes the dets into the normal regime, however a few hundred mK is much more that we usually see. Usually if we see this much heating, it is a sign that there is a bias short somewhere, and its dumping power somewhere it shouldn't be. Can you test one bg at a time to see if this is being caused by a single bias-line? If this is the case you may need to remove that bias-line from `cfg.dev.exp['active_bgs']` and avoid putting bias on it all-together.

yaqiongl, 2023-10-31:
Hi Jack, 

We are testing LP4 so our overbiasing voltage is only 18V

The thing I found confusing is that running` S.set_tes_bias_bipolar_array([0,18,18,18,0,18,0,0,0,0,0,0])` at low current mode has no heating issue at all

and if using `set_current_mode(S, bias_groups, 1, const_current=False) that pulls out overbias_voltage=18  from dev config, wouldcause heating. 

In both cases, they are using bias_groups=1,2,3,5

jlashner, 2023-10-31:
understood, I'm definitely support of adding a cfg option `overbias_using_high_current` [here](https://github.com/simonsobs/sodetlib/blob/c7bc19793ccfd168435885930d819fb58a506e8e/sodetlib/det_config.py#L105C1-L105C30) that allows you to configure whether you wish to use high-current-mode or not. Is that something you think you'd be able to add to the cfg and to the overbias_dets function?