simonsobs/sodetlib #414: uxm_relock does not use tone_power info in dev_cfg

msilvafe, 2024-02-01:
https://github.com/simonsobs/sodetlib/blob/32f4a8c5bdf545ca41efbb4aa7ba9a2addec0589/sodetlib/operations/uxm_relock.py#L45C1-L56C1

``uxm_relock`` needs to pass the tone_power in the dev_cfg to ``S.setup_notches`` or ``S.relock``.

jlashner, 2024-02-01:
I'm pretty sure that this line sets the default tone powers that are used for each band: https://github.com/simonsobs/sodetlib/blob/32f4a8c5bdf545ca41efbb4aa7ba9a2addec0589/sodetlib/operations/uxm_relock.py#L327

so when it runs reload_tune, it is using the tone-power in the device cfg. We should check that this is the case though, and probably allow for setting tone-power via kwarg to the `reload_tune` function.

msilvafe, 2024-02-01:
Ya it changed that variable in the `SmurfControl` name space but when I looked at the `S.get_amplitude_scale_array(4)` after relock was done everything was still at `tone_power = 12` despite `cfg.dev.bands[4]['tone_power']` being 9 and `S.amplitud_scale[4]` being 9