simonsobs/sodetlib #416: Fix responsivity in iv

satoru99, 2024-02-07:
There are two issues.
**1. Calculation of R_L**
The load resistance (R_L) should be shunt resistance (R_sh) + parasitic resistance (R_par) as written in comment.
It is calculated as follows
https://github.com/simonsobs/sodetlib/blob/1b384e513d3d4a3d4bfd8282f378b29e065c591b/sodetlib/operations/iv.py#L395-L397
However, from the equation of voltage:
`R_sh * I_bias = (R_tes + R_sh + R_par) * I_tes`.
the `R` evaluated here should be R_tes + R_par,
and, the `R_L` only contains R_par and not R_sh.
(Also, `R_n` might include contribution from R_par.)
To give the expected R_L, we should add `+ R_sh`.

**2. Calculation of dV/dI**
The responsivity is calculated as follows. Note that the `rL` comes from `R_L` above and thus underestimated by R_sh.
https://github.com/simonsobs/sodetlib/blob/1b384e513d3d4a3d4bfd8282f378b29e065c591b/sodetlib/operations/iv.py#L299
Here, dv is calculated from `v_tes`.
But from the theoretical model, the correct value seems to be the derivative of `v_bias`.
Since, `v_tes = v_bias * R / (R + R_L)`, this difference affects at small R. 

These two issues affect the responsivity calculation. It's necessary to be fixed.

msilvafe, 2024-02-07:
Thanks @satoru99, yes we (I) seem to have dropped `R_par` completely from all of these equations and we did not use the thevenin equivalent voltage (`i_bias*R_sh`) as you pointed out here. I follow your math/logic (from Irwin-Hilton) and agree that we should fix this. These seem like the are pretty simple changes to make, and I would love to see things tightening up at small R from this. Are you planning to show some example re-analyzed IVs and submit a PR with these fixes or work with @RemingtonGerras to do this?

RemingtonGerras, 2024-04-04:
[Confluence page of notes + summary of the PR](https://simonsobs.atlassian.net/wiki/spaces/~638a5b0b8fd2d2d5f13165d5/pages/374505518/Detector+Calibration+using+RP+Curve)