simonsobs/sodetlib #433: Relock Analysis Showing Large df p2p values while readout noise looks normal

samdayweiss, 2024-08-04:
HI all, a few days ago (with no clear origin), one of our wafers started showing a very low number for "good tracking channels", shown in the relock analysis plots (very large df p2p, see plots below). Despite this, readout noise looks quite reasonable. Possible that flux ramp drive parameters are off? Some help interpreting this would be great-- IV curves and bias steps for this UFM look great (it is Mv12 on srv SO4 (Mv12 is the photo with 411 good channels).

Although the other UFMs show a higher percentage in the green "good" range, several others show similar columns extending into high df p2p values.

@msilvafe @jlashner Max asked me to send this to you both :)

Kyohei Yamada first found this issue and can comment more on how it started.

[image]
[image]
[image]

jlashner, 2024-08-06:
Hi Sam, I believe we've generally increased the df_ptp allowed range in our device cfgs from 50 to 100 (here: https://github.com/simonsobs/sodetlib/blob/26b1e7ab02b21b797d8a618bf68217baed5fda42/sodetlib/det_config.py#L76), but in these plots it looks like the cutoff is 50 kHz (the old value).

We've changed this in the defaults, so new device cfgs should use that larger range. Any chance the device cfgs you're using are copied from an older version? If so I think you should just be able to update them such that `df_ptp_range=[0, 100]`.

samdayweiss, 2024-08-09:
Thanks Jack, I actually just checked the det_config files and it looks like the range is already set to [0,100]. How do we change the plotting to reflect this?

samdayweiss, 2024-08-09:
oh I see, I was confused, sorry I was looking at the sodetlib/sodetlib/det_config.py file. I'll change those device configs now.

Thanks!

jlashner, 2024-08-09:
Ah ya I can see how that's confusing. Hope that works!

jlashner, 2024-08-09:
I'm closing this, but if the problem isn't fixed feel free to re-open.

closed 2024-08-09 by jlashner