simonsobs/sodetlib #466: Extra and missing dependencies

BrianJKoopman, 2025-06-20:
I was looking at modernizing the packaging here and was wondering about some of the dependencies. `requirements.txt` currently lists (sorted):
```
ipykernel>=5.5.4
jedi==0.17.1
jupyterlab
lmfit
matplotlib
numpy
pandas
pyfftw
PyYAML
scipy
seaborn
setuptools >= 56.2.0
tqdm
```

The only dependencies here that I can find actually imported within any of the sodetlib code are:
```
lmfit
matplotlib
numpy
pandas
PyYAML
scipy
tqdm
```

That leaves these as seemingly "extra" dependencies:
```
ipykernel>=5.5.4
jedi==0.17.1
jupyterlab
pyfftw
seaborn
setuptools >= 56.2.0
```

I think we could pretty easily remove `jedi`, `seaborn`, and `pyfftw`. `setuptools` is maybe for packaging and would get dropped when I change how the packaging works. I'm not sure about `jupyterlab` and `ipykernel`. Presumably these were just getting installed because people use jupyter to look at things/interact with SMuRF? But they're not actually dependencies for the actual `sodetlib` code.

My question here is do we want to keep them around in some separate `requirements.txt` file? Or can I just drop them and assume people will install jupyter themselves if they want it?

### sotodlib
My other question relates to `sotodlib`, which is imported several places within `sodetlib`. In some places it's imported within a function, but other places it's at the top of the file. Since this isn't pulled in automatically, there are often instances of `sodetlib` being installed, but not actually importable because it hits a `ModuleNotFoundError`.

In the past I think there was resistance to adding `sotodlib` as an explicit dependency. Do we still want to avoid that?

My preferences are either to add it to the `requirements.txt` file as:
```
sotodlib @ git+https://github.com/simonsobs/sotodlib.git@master
```

This will install the latest `master` at the time of installation. Or we could pin to a specific commit.

Or, to protect all the imports within `sodetlib`. I don't know how often the functions that use these imports are called/used and in what context. If I just move all the imports into the functions that use them, is that fine? Should users be warned they're missing an (optional?) dependency?

msilvafe, 2025-06-23:
I'm ok trying this out but I would want to test it for a bit by running one of the platforms on this branch for a little bit because I don't really know what all of these dependencies are used for.

tristpinsm, 2025-11-20:
added requirements that looked relevant [here](https://github.com/simonsobs/smurf_dockers/commit/d927715aae223348a14da4f87ef24100eb2bea06)

BrianJKoopman, 2025-11-20:
> added requirements that looked relevant [here](https://github.com/simonsobs/smurf_dockers/commit/d927715aae223348a14da4f87ef24100eb2bea06)

Great, both of those actually get pulled in by the `jupyter` dependency that is already there. See https://pypi.org/project/jupyter/ -- "pip install jupyter installs the Jupyter Notebook, JupyterLab, and the IPython Kernel."

I dropped most of the 'extra' dependencies I listed above in #483.